OpenOCL/OpenOCL #123: Write a parser for .ocml

jkoendev, 2019-06-10:
First draft of the language is presented here:
https://www.syscop.de/event/introduction-ocml-optimal-control-modeling-language

We might also want a parser in a standalone language like C++, python or javascript

.ocml github project (not up to date):
https://github.com/OpenOCL/ocml

jkoendev, 2019-06-10:
Could parse .ocml by any langauge (prob. python is prefered) and turn into a easier computer readable format like `json`

grigorem, 2019-12-13:
I think I can make it directly in Matlab but I want from you a concrete example of how do you want the output (struct, cell array, object).
Also, do you want to parse the formulas as well? Do you want the numbers evaluated or as strings?