OpenOCL/OpenOCL #129: Time Varying Linear Quadratic control

mishapadidar, 2019-07-18:
I'm trying to solve a Linear Quadratic optimal control problem with time varying matrices. 

Question 1:
Is there a way to pass a time variable into the  differential equation handler so I can construct these _time-varying_ matrices? For example my differential equation might be
`
dx/dt = A(t)*x + B(t)*u
`
where the matrices A(t), B(t) are
`
A(t) = [t 0 ; 0 t^2]

B(t) = [1 0; 0 2*t]
`

Question 2:
Is there a way to write the differential equations and costs in matrix form? So far I have been writing out the differential equations one by one as in the following
` daeh.setODE('x1', u.u1*cos(x.x3));
  daeh.setODE('x2', u.u1*sin(x.x3));
`

jkoendev, 2019-07-22:
Hi, sorry for the late reply!

q1: 
you can add a state for the time, e.g. as in the cartpole example:
```varh.addState('time', 'lb', 0, 'ub', 10);```
and differential equation:
```daeh.setODE('time', 1)```
[cartpole.m](https://github.com/OpenOCL/OpenOCL/blob/master/%2Bocl/%2Bexamples/cartpole.m)

q2:
you can add a matrix valued variable, e.g.
```varh.addState('x', [5,4])```
and differential equation in matrix form
```daeh.setODE('x', A*x+B*u)```

There is an example in the current `dev_acados` branch:
[mass_spring](https://github.com/OpenOCL/OpenOCL/tree/dev_acados/%2Bocl/%2Bexamples/%2Bmass_spring)

jkoendev, 2019-07-22:
for the time you probably also need to set a bound on the initial time:
```
solver.setInitialBounds('time', 0);
```

mishapadidar, 2019-07-22:
Thanks for the support very helpful.

closed 2019-07-22 by mishapadidar