OpenOCL/OpenOCL #130: reference array via a state value

mishapadidar, 2019-07-22:
We have a quadratic cost `x'*A*x` where `A` is a function of some time series data. In order to build this time dependent matrix we need to be able to retrieve the data from some time variable,
i.e.
data(state.time) 
or
interpolate the data based on state.time.value.

minimal example:
Given data 
`z= [1.11 3.4 2 4.3 7.9 12.1 ...] ` 
we want to build a matrix A where at integer times t, 
`A(t) = [ z(t)  z(t+1); z(t-1)  z(t)]`

Is there a way to use a time variable in such a way. As far as I have tested It seems that the states don't have an assigned numerical value until the solver completes, so this may be impossible.

jkoendev, 2019-07-22:
I think you can implement this by specifying `gridcosts` (see [docs of Solver](https://openocl.org/api-docs/#apiocl_solver))

You implement a function with the signature
```
function gridcosts(cost, k, K, x, p)
```

If you have N control intervals, the state trajectory is represented by states at `K=N+1` gridpoints (or timepoints). The `gridcosts` function will be called K times with the indizes `k` from 1 to K for the current gridpoint (k and K are numeric values, integers). 

If you have a fixed endtime `T` your current time is `t=k/K*T`.

If you have a free endtime, things get trickier. Let me know if that is the case, possible solutions would be time transformation or using polynomial interpolation/splines etc.

jkoendev, 2019-07-22:
sorry in the current version v5.08 those cost terms are called `pointcosts`. I will try to make a release tomorrow to avoid confusion, i changed the name to gridcosts in the current master branch and in the docs.

mishapadidar, 2019-07-22:
This is what I was looking for. Is there a way to include control variables into the grid cost?

jkoendev, 2019-07-23:
Yes `x` are the states only. 
You can declare your controls as states, and introduce new controls `u_prime` and define the differential equations
```
daeh.setODE('u', u_prime)
```

so u_prime are the time derivatives of your (original) controls.

jkoendev, 2019-07-23:
i made a new release `v5.11` you can get it from [here](https://openocl.org/get-started/) or clone the master. Let me know if you have any questions

closed 2019-07-24 by mishapadidar