OpenOCL/OpenOCL #146: could setParameter work with struct

pbpf, 2019-09-25:
Think about there are many parameters, so that it will be too bad to add them one by one. So I trying to do something like this 
>------------------
earth.R0=6378137;
earth.g0=9.8;
p.earth=earth;
p.x0=1;
solver.setParameter('p'   , p);
--------
and in varsfun
>----------------------
sh.addParameter('p');
----
Then in odefun
I want to get it back.

jkoendev, 2019-09-29:
If your parameters are just constant values then it is better not to use parameters but pass the values using anonymous functions. You can see how to do it in this example:
https://github.com/OpenOCL/OpenOCL/blob/master/%2Bocl/%2Bexamples/ballandbeam.m

All the constants are stored in a struct `conf`. You can pass a anonymous function to the solver like (for the system synamics):
```
@(sh,x,z,u,p) daefun(sh,x,z,u,p,conf);
```

and so daefun has `conf` as an input.

You can also use the parameters but it will be not efficient. And you can not use struct, but add them one by one:
```
sh.addParameter('p')
sh.addParameter('R0')
etc.
```

jkoendev, 2019-10-30:
note that we introduced the `userdata` field to pass constant parameters/values/structs..

See:
https://github.com/OpenOCL/OpenOCL/blob/v7.03/%2Bocl/%2Bexamples/ballandbeam.m

closed 2019-10-30 by jkoendev