OpenOCL/OpenOCL #148: interpolation

kmg232, 2019-10-02:
I presume the CasADi interpolant can already be used in OpenOCL? If yes, would it be possible to introduce an example where lookup tables are used in the definition of the dynamic system (dae or ode) and path constraints?

jkoendev, 2019-10-02:
I think interpolant should work although i haven't tested it yet. 
You need to set the option `nlp_casadi_mx` to `true` for `ocl.Solver` (which will be `ocl.Problem` in the next release) because interpolant only works with MX symbolics as far as i know.

Generally you can use all CasADi functions with OpenOCL. Sometimes it can happen that the OpenOCL type (`ocl.Variable`) and CasADi types do not work well with each other, then you can always use the `.value()` function on the `ocl.Variable` to get the underlying CasADi value and work with that.

I assume your lookup table is state dependent? What are you interpolating in your case?
I will try to come up with a simple example. 

If polynomial interpolation is sufficient for you problem, you could avoid using interpolant by using `polyfit` to fit your data, and then `polyval` in your OpenOCL models. Would that work for you?

kmg232, 2019-10-02:
Yes the lookup table I would like to use is either state or control dependent. Taking your race car example, you could have a lookup table for the road center line e.g. y_center = f(x_road).

jkoendev, 2019-10-04:
hi i added an examples racecar_interpolant here:
https://github.com/OpenOCL/OpenOCL/blob/dev_interpolant_example/%2Bocl/%2Bexamples/racecar_interpolant.m

It should show the basic usage of casadi.interpolant in OpenOCL.
Note that in line 163 i use the `.value` to get the casadi type. In line 164 and 165 i add the centerline (casadi expression) from the right side so that it works with the `ocl.Variable` type. Otherwise you could also use the `.value` function on the other part of the expression like:
```
y_max = y_center + 0.5*p.road_bound.value;
y_min = y_center - 0.5*p.road_bound.value;
```

It is in a seperate branch as i haven't written any test for it yet..