OpenOCL/OpenOCL #154: Add random disturbance

ChahidAbderrazak, 2020-05-04:
Dear,
I am a beginner in OCL and I found you software very usefull for my nonlinear model. However, 
I am trying to add random  gaussian noise to **ocl.problem** to check the robustness of  the optimal control. But, I could not succeed!!!.  (I am using the carepool template for my system)
Could you please add documentation or an example about it?
Thank you very much.
Abderrazak

jkoendev, 2020-05-07:
Hi Abderrazak, there is no direct support for random variables in OCL. Unfortunatly i am also not very familiar with "robust" or "stochastic" optimal control. To get an idead of the robustness w.r.t. some of the parameters, you could make sweep over the parameters, so you change a parameter and recalculate the solution.

ChahidAbderrazak, 2020-05-09:
Thank you very much for your suggestion. 
I will check it out :):).