OpenOCL/OpenOCL #155: Initial guess constraint violation

Foxp1, 2020-05-15:
Hi,
I'm setting an initial guess for the optimization with problem.initialize, and even though the constraints should be satisfied, the "constraint violation" output is rather high (I provide values for state and control for every grid point with problem.initialize). Any suggestions on how to check if the issue comes from the system dynamics constraints? Thanks!

jkoendev, 2020-05-17:
Hi, there is no direct way at the moment to evaluate the constraints function for the different type of constraints. You could try to evaluate the (casadi) constraints function on the initial guess and see if you can figure out which constraint are violated. For this you can set a breakpoint in `+ocl/+casadi/CasadiSolver.m` around line 346 before this line which calls the solver:
```
sol = self.nlpData.solver.call(args);
```
Then create a CasaADi function and evaluaste with the initial guess
```
g = casadi.Function('g', {self.nlpData.casadiNLP.x}, {self.nlpData.casadiNLP.g})
constraints = full(g(v0));

figure; hold on; grid minor;
lb = self.nlpData.constraints_LB;
ub = self.nlpData.constraints_UB;
lb(isinf(lb)) = min(constraints );
ub(isinf(ub)) = max(constraints);
stairs(lb, 'g')
stairs(ub, 'r')

stairs(constraints, 'k--', 'LineWidth',1)
```

ill post an example in a minute

jkoendev, 2020-05-17:
For the Racecar example it looks like this:

[image]

For each control interval, there are in this order
- 4 grid constraints (yellow)
- 21 dynamics constraints for d=3 and n_x=7 (green)
- 7 parameters (blue)
- 1 constraint for the time (red)
- 7 continuity constraints, because number of state n_x=7 (purple)

Red and green lines are upper and lower bounds for the constraints. The constraints are in the form g<=0 or g==0, so the upper bound is always 0 and the lower bound either 0 or minus infinity.

Foxp1, 2020-05-17:
This is very helpful -- thanks very much!
I noticed that the user-provided control values are not assigned in getInitialGuessWithUserData.m. After changing that the obtained solution is as expected -- although I still can't figure out why the constraint violation is rather large at the initial step. What is the collocation method used in openOCL (e.g. trapezoidal or Hermite-Simpson?)

jkoendev, 2020-05-17:
ah yes the controls were missing when using .initialize(), if you fixed you could do a pull request, or i will fix it in the coming days.. 
The collocation method is based on Lagrange polynomials at Radau points.

Foxp1, 2020-05-17:
Great, thanks for the info! I just did a pull request as well.